spapadim/argon1 #14: FAN is not turned on for V2 case (normal HDMI)

sergiuko83, 2021-02-19:
Not exactly sure why this is reversed, but it only works like this in my case...
  def _bus_write(self, value: int, register: int = _SMBUS_REGISTER):
    # Could raise IOError, according to "official" scripts
    self._bus.write_byte_data(_SMBUS_ADDRESS, int(value), register)

The initial version is sending 0x00 (register value which is 0xd3 in my case but it seems to be irrelevant) which turns off the fan.
The I2C command for argon case is
i2cset -y _SMBUS_DEV _SMBUS_ADDRESS value
i2cset -y 1 0x1a 0x00
...
i2cset -y 1 0x1a 0x64

Command guide: https://github.com/Argon40Tech/Argon-ONE-i2c-Codes

Are you sure the initial version spins the fan?
The display part is working without problems, IO errors not returned, but fan does not turn at all

marcelogaio, 2021-04-06:
Same here! Tried it with 0x1a and it works...

alexsotoaguilera, 2021-04-06:
how can I check if my fan is working?

marcelogaio, 2021-04-06:
> how can I check if my fan is working?

Sound? ;) It's not super loud bu you definitely can hear it if it's working.
Otherwise, open the case and you can see it inside.

marcelogaio, 2021-04-06:
> > how can I check if my fan is working?
> 
> Sound? ;) It's not super loud but you definitely can hear it if it's working.
> Otherwise, open the case and you can see it inside.

Also: try this command to see if it works:
`i2cset -y 1 0x1a 0x64`

alexsotoaguilera, 2021-04-06:
the fan turns on with command:
'''
i2cset -y 1 0x1a 0x64
'''
and when pi hits temperature below 50 degrees stop the loud. I dont sure if fan stops or still working at low revolutions.

sergiuko83, 2021-04-07:
In my experience, even at lowest FAN speed you can still hear it.
The 0x64 is the percentage of speed meaning 100%
You can experiment with lower values, for example 0x01 for 1%, 0x0a for 10%, 0x32 for 50% etc

I am not sure if this reversal of arguments is only for the V2 cases (large HDMI) or it applies to V1 (HDMI mini) as well

alexsotoaguilera, 2021-04-07:
I confirm this behaviour, it only turns on the fan with the command. The argon1 service need a fix. My cases is the v2 version.

webbes, 2021-04-07:
Exactly the same issue here:

`i2cset -y 1 0x1a 0x64
`

Works perfectly fine...otherwise the fan never turns on...

MrMrRobat, 2021-04-25:
@sergiuko83, thanks, the patch with swapping arguments worked like a charm.

In my case path to the file was `/usr/lib/python3/dist-packages/argonone/__init__.py`